RoboMaster/RoboRTS #7: make error in "teb_optimal.cpp"

dxls168, 2018-02-01:
hi：
    When I tried to compile the whole project, I found the problem was initialization.
    I tried to mask some code, the compiler passed, but I do not know how to amend mistakes. I used the latest version of g2o.
    Looking forward to your reply!

[ 92%] Building CXX object modules/planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/teb_optimal.cpp.o
/home/ls/my/robomaster/RoboRTS/modules/planning/local_planner/timed_elastic_band/teb_optimal.cpp: In member function ‘boost::shared_ptr<g2o::SparseOptimizer> rrts::planning::local_planner::TebOptimal::InitOptimizer()’:
/home/ls/my/robomaster/RoboRTS/modules/planning/local_planner/timed_elastic_band/teb_optimal.cpp:140:64: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >::BlockSolver(rrts::planning::local_planner::TebLinearSolver*&)’
   TebBlockSolver* blockSolver = new TebBlockSolver(linearSolver);

[image]

KevinLADLee, 2018-02-01:
We use the following version g2o. FYI.
“ros-kinetic-libg2o/xenial,now 2016.4.24-0xenial-20171104-173138-0800 amd64 ”
Please provide your current version of g2o.

dxls168, 2018-02-01:
hi:
  I try to reinstall g2o with the command "sudo apt-get install ros-kinetic-libg2o" and project compile all goes well.
 Thank you very much for your reply!

closed 2018-02-01 by dxls168