RoboMaster/RoboRTS #14: Local Planner Dosen't Work

Robin970822, 2018-03-20:
In the newest RoboRTS, local planner does not work properly. Log information is shown below. When I reset to previos commit, it works again. 
[libprotobuf ERROR google/protobuf/text_format.cc:274] Error parsing text-format rrts.planning.local_planner.Config: 48:32: Need space between number and identifier.
[libprotobuf ERROR google/protobuf/text_format.cc:274] Error parsing text-format rrts.planning.local_planner.Config: 49:5: Message type "rrts.planning.local_planner.Obstacles" has no field named "maibva".

dongleecsu, 2018-03-20:
As the error message says, in the [teb config file](https://github.com/RoboMaster/RoboRTS/blob/master/modules/planning/local_planner/timed_elastic_band/config/timed_elastic_band.prototxt#L47), just change `obstacle_poses_affected: 30maibva` to `obstacle_poses_affected: 30`.

Robin970822, 2018-03-20:
Got it

closed 2018-03-21 by Robin970822