RoboMaster/RoboRTS #18: 串口模块仍然无法与开源固件传输数据

Mor-harry, 2018-03-29:
双向测试发现，都可以正常发送数据，但是接受过程中的解码似乎出现问题，主要体现是串口助手读取车或者tx2端都有数据，但是互相解码的过程中无法正常解码，导致两边系统无法通信

Robin970822, 2018-03-29:
我们遇到相同的问题！但是rm端无法读取命令码为0x0001到0x0005的数据。（这些数据是由裁判系统产生，我们测试了裁判系统的端口，发现有这些信息发出，但不是按照手册上的通讯协议的格式，比如命令码为0x0001的数据长度为35而非手册中的25；然而我们的确发现裁判系统发出的信息部分包含通讯协议中我们所需要的信息，比如命令码为0x0001的数据中的确有当前血量的信息而没有当前最大血量的信息，命令码为0x0002的数据中的确有哪个装甲板受攻击的信息，而没有受伤害种类的信息。）

KevinLADLee, 2018-03-30:
这是因为裁判系统主控在最新的固件中更新了协议，我们测试通过后会及时更新RoboRTS和RoboRTS-Firmware。
Sorry about that. The referee system protocol has been changed. We will update our code in RoboRTS and RoboRTS-Firmware in next commit.

eeluobu, 2018-03-30:
The latest protocol in [RoboRTS protocol](https://github.com/RoboMaster/RoboRTS-Firmware/blob/master/Doc/protocol/readme.md) has been updated, please update the latest RoboRTS-Firmware, and update the Referee-System firmware by [RoboMaster Assistant](https://www.robomaster.com/zh-CN/products/components/assistant)

commit 9baf6240 (2018-03-30): Fix #18
closed 2018-03-30 by noah003