RoboMaster/RoboRTS #23: [Error] Install RoboRTS in TX2  :cannot find -lgxiapi

Wangergou123, 2018-07-29:
Hello ,I install RoboRTS in TX2, when i catkin_make,it show this error: 
 /usr/bin/ld: cannot find -lgxiapi
collect2: error: ld returned 1 exit status
RoboRTS-master/modules/driver/camera/mercure/CMakeFiles/mercure.dir/build.make:165: recipe for target '/home/nvidia/RoboRTS_ros/devel/lib/libmercure.so' failed
make[2]: *** [/home/nvidia/RoboRTS_ros/devel/lib/libmercure.so] Error 1
CMakeFiles/Makefile2:6455: recipe for target 'RoboRTS-master/modules/driver/camera/mercure/CMakeFiles/mercure.dir/all' failed
make[1]: *** [RoboRTS-master/modules/driver/camera/mercure/CMakeFiles/mercure.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

KevinLADLee, 2018-07-30:
"libmercure.so" is a camera driver we used. If you do not use this camera on Jetson TX2, you could comment out following lines to build without it:
modules/driver/camera/CMakeLists.txt line 3-8, 43-48, 62-67
modules/driver/camera/camera_node.h line 30-32

Wangergou123, 2018-07-30:
Thank you so much ,this error has been solved with your help !

closed 2018-07-30 by Wangergou123