RoboMaster/RoboRTS #25: is Yolo module working properly?

MMMHT, 2018-09-08:
Hi
I am lucky to have partipated the 2018 ICRA AI Challenge. and was impressive about DJI autoRobot.
I wonder if the robot is detected using Yolo module?cause recently I also using Yolo to detect object but was failed to get yolo network result currectly. the weight is trained by myself and was checked by using ./darknet ....... ,everything looks fine,after I encapsulation Yolo srource file in the way RoboRTS do,and run in C++,even the image I passed into yolo is absolutly right ,I still cant get currect yolo detection result. the object number yolo returned me is always 0,and the windows yolo drawed by itself is also no reaction.hopes anyone can helps me to figure out what is going wrong .

noah003, 2018-09-10:
Please confirm the camera_node is work properly and the .cfg file of inference and training is the same one

MMMHT, 2018-09-17:
@noah003 thanks very much ，I finally solved this problem by replace the image.c file,and it work all fine.