RoboMaster/RoboRTS #26: Install RoboRTS in NUC: missed file

DeriZSY, 2018-11-09:
fatal error when "catkin_make":
modules/perception/detection/armor_detection/constraint/set/proto/constraint_set.pb.h no such file or directory

DeriZSY, 2018-11-09:
"[Warning] message is not found in workspace " when "catkin_make messages_generate_messages".
Cannot find any messages implemented in RoboRTS by "rosmsg list"
invalid message type when trying to publish EnemyPos using  "rostopic pub"

DeriZSY, 2018-11-20:
> Most of the errors that lead to the fatal errors do cause due to not running catkin_make messages_generate_messages BEFORE running catkin_make, as instructed on the documentation.
> For the message or service issue, RoboRTS' built messages are NOT a ROS message type, but a header file. As ROS message files end with .msg and the files used in RoboRTS ends with .h, it won't show up on the rosmsg list command.

1. For the first problem, we are pretty sure we have run catkin_make_message_generate_messages before catkin_make, but there are still missed file error, and this can only be solved by running catkin_make multiple times. I've had all the people in the algorithm group of our team installed the RoboRTS, but every one meets exactly the same problem. We are not that silly to skip steps in the guide.
2. All topics using message types defined in RoboRTS all died automatically. What's more, when we tried to manually publish a message of message type defined in RoboRTS (for example when we tried to publish a \EnemyPos message to test the function of gimbal), it says 'Message Invalid'. 
By the way, messages in RoboRTS are also defined in .msg file, in "RoboRTS/modules/stream/messages/msg/" folder. As far as I know, for all messages in ROS, you have to first write a .msg file, and then compile it to .h file.

senyig20, 2018-11-20:
That's weird, I've been testing it on multiple platforms including AWS, Azure and local computers and only error I got was some lib only compatible w/arm, but I got that only on x86&64 machines, and all arm machines compiled as it's supposed to. 

If you want to try with an older version which has been tested on an Jetson, I have a fork of that, you can try using that as well.

senyig20, 2018-11-20:
See my pull request for a fix for the message issue.

DeriZSY, 2018-11-20:
> See my pull request for a fix for the message issue. (If needed)

Thanks , I'll check it

0aqz0, 2018-12-04:
You could generate the protobuf files (*.pb.h, *pb.cc) by yourself.
For example, in the directory ~/roborts_ws/src/RoboRTS/modules/perception/detection/armor_detection/constraint_set/proto
```bash
$ protoc -I=. --cpp_out=. ./constraint_set.proto
```
Then you should find constraint_set.pb.h and constraint_set.pb.cc.

closed 2019-01-24 by KevinLADLee