RoboMaster/RoboRTS #28: error This file was generated by a newer version of protoc

0aqz0, 2018-12-03:
catkin_make failed
```bash
/home/haodong/roborts_ws/src/RoboRTS/modules/driver/camera/proto/camera_param.pb.h:12:2: error: #error This file was generated by a newer version of protoc which is
 #error This file was generated by a newer version of protoc which is
  ^
/home/haodong/roborts_ws/src/RoboRTS/modules/driver/camera/proto/camera_param.pb.h:13:2: error: #error incompatible with your Protocol Buffer headers. Please update
 #error incompatible with your Protocol Buffer headers.  Please update
  ^
/home/haodong/roborts_ws/src/RoboRTS/modules/driver/camera/proto/camera_param.pb.h:14:2: error: #error your headers.
 #error your headers.
```
my protobuf version:
2.6.1-1.3

KevinLADLee, 2018-12-04:
You could use this command to delete all proto files generated before and re-compile RoboRTS.

`find ${ROBORTS_PATH} -name "*.pb.h" -o -name "*.pb.cc" | xargs -i mv {} /tmp/ 
`

0aqz0, 2018-12-04:
The problem is that I have installed protobuf2.6 and protobuf3.0. And the default version is 3.0.
I solved it by setting the default version to 2.6.