RoboMaster/RoboRTS #33: some mistake in motion model

lyziew, 2019-04-02:
Accroding to https://github.com/ros-planning/navigation/issues/499
It seem should be:
```
  double trans_hat_stddev = (alpha3 * (delta_trans*delta_trans) +
                             alpha4 * (delta_rot*delta_rot));,
  double rot_hat_stddev = (alpha1 * (delta_rot*delta_rot) + 
                           alpha2 * (delta_trans*delta_trans));
  double strafe_hat_stddev = (alpha4 * (delta_rot*delta_rot) + 
                              alpha5 * (delta_trans*delta_trans));
```

lyziew, 2019-04-02:
Also in referee_system.cpp  RefereeSystem::RobotHeatCallback
heat value is not used
may should be
```
  roborts_msgs::RobotHeat robot_heat;
  robot_heat.chassis_volt = raw_robot_heat->chassis_volt;
  robot_heat.chassis_current = raw_robot_heat->chassis_current;
  robot_heat.chassis_power = raw_robot_heat->chassis_power;
  robot_heat.chassis_power_buffer = raw_robot_heat->chassis_power_buffer;
  robot_heat.shooter_heat = raw_robot_heat->shooter_heat0;
  ros_robot_heat_pub_.publish(robot_heat);
```

charmyoung, 2019-04-03:
We have fixed the latter one. 
According to the omni odom model, we will take it into consideration for the next release version after the competition maybe.

closed 2019-04-03 by lyziew