RoboMaster/RoboRTS #37: catkin_make error

wangshulei, 2019-09-25:
There some errors when I make the project，like the picture, can you tell me what's wrong? My Tx2 environment is ubuntu 18.04 and ROS melodic.
[image]

XinZhangk, 2019-09-25:
This code base is prepared to run on ubuntu 16 + kinetic. There are a few updates on the syntax of some libraries between kinetic and melodic. You can circumvent the incompatibility by finding the culprits and manually changing them.

KevinLADLee, 2019-12-29:
Hi, @wangshulei! If your environment is Ubuntu 18.04, you could refer to my [RoboRTS-Edu](https://github.com/Artinx-Robotics/RoboRTS-Edu) project forked from this RoboRTS project. 

I have fixed some dependencies and build errors on Ubuntu 18.04 (Jetson Nano). Hope this will be helpful to you.

closed 2020-01-06 by KevinLADLee