RoboMaster/RoboRTS #38: can you tell me how to deal with this problem?

cd4806088, 2019-10-27:
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
roborts_bringup: Cannot locate rosdep definition for [roborts_tf]

cd4806088, 2019-10-27:
can you tell me how to deal with it?

XinZhangk, 2019-10-27:
With the little information you have provided I can only speculate. But it looks to me that you didn't finish a standard ros installation. Have you run the following commands?
```bash
sudo rosdep init
rosdep update
```

KevinLADLee, 2019-12-29:
The package [roborts_tf] is deprecated. You could delete it in 'roborts_bringup/package.xml' .

https://github.com/RoboMaster/RoboRTS/blob/ros/roborts_bringup/package.xml#L20

cd4806088, 2019-12-30:
thank you, got it------------------ 原始邮件 ------------------

 
https://github.com/RoboMaster/RoboRTS/blob/ros/roborts_bringup/package.xml#L20
 
&mdash;

closed 2020-01-03 by KevinLADLee