RoboMaster/RoboRTS #39: 重载地图后定位问题

linyufeishi, 2019-11-12:
用gmapping建立地图后（地图文件包括一个pgm图片he一个描述该图片的元数据的yaml文件），用您提供的模块组织了一个launch文件，这个launch文件主要是用来重载之前建立的地图，并在该地图上进行导航，launch文件内容如下：
[image]

但是运行这个launch文件后，通过rviz，可以观察到如下的情景：Baselink坐标系在重载的地图上一直不停地大范围跳动。我认为这可能是定位不准的问题，但是不知道如何解决，期望能得到您的帮助，谢谢。

2B-people, 2019-11-13:
你好，我用过这个框架，你这个问题我猜是在amcl定位的问题，你检查下你的amcl节点（你的launch上为localization_node）可以用rqt 看下是否启动了这个节点，或者这个节点是否报错

 

 <https://user-images.githubusercontent.com/33945028/68636498-9f777380-0536-11ea-881e-192142644b42.PNG>

linyufeishi, 2019-11-13:
我把launch文件修改了下，去掉了static transform between /odom and /map，因为后来我查询资料发现amcl也会发送这个topic，且实际应用过程/odom 和/map之间的变换关系也是变化而不是静态的。这样之后，baselink跳动的频率小了很多，用rviz的 2D pose estimate给小车初始定位后，小车停个几秒钟，位置又漂了。想问下amcl是不是小车不运动，停下来后定位就不准了呀？

2B-people, 2019-11-13:
原来你发布了一个静态的tf啊baselink和odom的应该的变化是静态的，你得看看你的tf_tree，，，，，我觉得不是吧，位置飘了你要一点点查，有可能是amcl，amcl是根据周围环境的外定位,在原地不动应该是不会变化的啊，是发布odom到map的tf变化，也有可能是在odom坐标系下你车子本身的odom数据有问题，体现在rviz上就是位置漂移几秒，又被矫正回来，这时候你的odom坐标系原点会变化。

可以加qq：996599620，我是学生，不一定能解决你的问题，但是欢迎交流😁

closed 2019-12-30 by KevinLADLee