RoboMaster/RoboRTS #40: catkin make error - tx2, ubuntu 16.04, kinetic

SteinwayCaiZhuo, 2019-12-28:
I was running catkin_make command to after I cloned the RoboRTS repo to /src directory and I got the following message:
`
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
RoboRTS/roborts_planning/local_planner/CMakeFiles/teb_test.dir/build.make:93: recipe for target 'RoboRTS/roborts_planning/local_planner/CMakeFiles/teb_test.dir/src/teb_test.cpp.o' failed
make[2]: *** [RoboRTS/roborts_planning/local_planner/CMakeFiles/teb_test.dir/src/teb_test.cpp.o] Error 4
`
The RoboRTS repo is up-to-date and I can successfully run catkin_make command in an empty workspace.

XinZhangk, 2019-12-28:
It would be more helpful if you you can post all messages. 
I think before teb_test local_planner node has to be built first, and I wonder if local_planner is successfully built. 
If so, then a quick fix is to simply remove the test executable for now: this particular executable only helps testing, RoboRTS can run without it.
If otherwise, have you tried to compile a simple ROS project on your system before? I would check the path settings first for odd errors like this.

SteinwayCaiZhuo, 2019-12-29:
I tried removing teb_test.cpp but still got a similar error. 
`c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
make[2]: *** [RoboRTS/roborts_planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/src/teb_local_planner.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
<command-line>:0:15: warning: ISO C++11 requires whitespace after the macro name
make[1]: *** [RoboRTS/roborts_planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
Invoking "make -j6 -l6" failed
`
The complete messages with and without teb_test are in attachment. 
[catkin_make_with_teb_test.txt](https://github.com/RoboMaster/RoboRTS/files/4007965/catkin_make_with_teb_test.txt)
[remove_teb_test.log](https://github.com/RoboMaster/RoboRTS/files/4007966/remove_teb_test.log)

I can successfully compile an empty ROS project.

KevinLADLee, 2019-12-29:
Hi! @SteinwayCaiZhuo. Could you please show me more about your computer hardware information? 

And I think you could try to increase your swap file or compile RoboRTS by command 'catkin_make -j1'.

SteinwayCaiZhuo, 2019-12-30:
@KevinLADLee Thank you a lot for your help. I tried catkin_make -j1 but still got the same error. Since I can use catkin_make -j6 in empty ros projects, multi-threading making should not be the problem. 

I guess the most possible reason is developing environment regarding cmake, gcc. However, I am not familiar with it. Could you show me what to check and what should the configurations be like? 

My device is Nvidia Jetson TX2. 

nvidia@tegra-ubuntu:~$ lscpu
Architecture:          aarch64
Byte Order:            Little Endian
CPU(s):                6
On-line CPU(s) list:   0-5
Thread(s) per core:    1
Core(s) per socket:    3
Socket(s):             2
Model name:            ARMv8 Processor rev 3 (v8l)
CPU max MHz:           2035.2000
CPU min MHz:           345.6000
L1d cache:             32K
L1i cache:             48K
L2 cache:              2048K

SteinwayCaiZhuo, 2019-12-30:
@XinZhangk  Thank you very much for your suggestion. I removed teb_test target in CMake and then the entire project can be built successfully. 

Yesterday I made the following comment. I was removing the teb_test.cpp itself instead of target teb_test from CMakeLists.txt. 

> I tried removing teb_test.cpp but still got a similar error.
> `c++: internal compiler error: Killed (program cc1plus) Please submit a full bug report, with preprocessed source if appropriate. See [...]

closed 2019-12-30 by SteinwayCaiZhuo