RoboMaster/RoboRTS #41: Can not open device: /dev/serial_sdk

SteinwayCaiZhuo, 2019-12-30:
When I ran command:
rosrun roborts_base roborts_base_node
or
roslaunch roborts_bringup base.launch

I got the following error:
WARNING: Logging before InitGoogleLogging() is written to STDERR
E1230 23:24:22.297278  9929 handle.cpp:29] Can not open device: /dev/serial_sdk. Please check if the USB device is inserted and connection is configured correctly!

Some information:
nvidia@tegra-ubuntu:~$ lsusb
Bus 002 Device 020: ID 8086:0b3a Intel Corp. 
Bus 002 Device 017: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 011: ID 0483:5740 STMicroelectronics STM32F407
Bus 001 Device 007: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 010: ID 17ef:602e Lenovo 
Bus 001 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 10c4:ea60 Cygnal Integrated Products, Inc. CP210x UART Bridge / myAVR mySmartUSB light
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

nvidia@tegra-ubuntu:~/robo_ws/src/RoboRTS/roborts_bringup$ cat scripts/udev/roborts.rules 
# set the udev rule
KERNEL=="ttyACM*", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", MODE:="0777", SYMLINK+="serial_sdk"
KERNEL=="ttyUSB*", ATTRS{idVendor}=="10c4", ATTRS{idProduct}=="ea60", MODE:="0777", SYMLINK+="rplidar"

Thank you!

closed 2020-01-01 by SteinwayCaiZhuo

iltlo, 2022-06-11:
I faced the same problem, may I know how did you fix that?

Freedom-Guo, 2022-06-11:
您的邮件我已经收到，假期愉快！