RoboMaster/RoboRTS #42: protocol.cpp sending timeout

SteinwayCaiZhuo, 2020-01-01:
When I ran
roslaunch roborts_bringup base.launch

I got:
`nvidia@tegra-ubuntu:~/robo_ws$  roslaunch roborts_bringup base.launch
... logging to /home/nvidia/.ros/log/85a77314-2c88-11ea-b9c4-00044bc60ac7/roslaunch-tegra-ubuntu-28819.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/nvidia/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

WARNING: ignoring defunct <master /> tag
started roslaunch server http://localhost:40396/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /serial_port: /dev/serial_sdk

NODES
  /
    roborts_base_node (roborts_base/roborts_base_node)

ROS_MASTER_URI=http://localhost:11311

process[roborts_base_node-1]: started with pid [28838]
WARNING: Logging before InitGoogleLogging() is written to STDERR
I0101 19:19:45.388367 28838 handle.cpp:33] Connection to /dev/serial_sdk
I0101 19:19:45.389498 28838 handle.cpp:39] Initialization of protocol layer and dispatch layer succeeded. 
E0101 19:19:45.449671 28850 protocol.cpp:102] Retry session 2
E0101 19:19:45.500548 28850 protocol.cpp:102] Retry session 2
E0101 19:19:45.552613 28850 protocol.cpp:102] Retry session 2
E0101 19:19:45.560545 28850 protocol.cpp:102] Retry session 3
E0101 19:19:45.604527 28850 protocol.cpp:102] Retry session 2
E0101 19:19:45.612335 28850 protocol.cpp:102] Retry session 3
E0101 19:19:45.656569 28850 protocol.cpp:98] Sending timeout, Free session 2
E0101 19:19:45.663347 28850 protocol.cpp:102] Retry session 3
E0101 19:19:45.714380 28850 protocol.cpp:102] Retry session 3
E0101 19:19:45.765506 28850 protocol.cpp:98] Sending timeout, Free session 3
`

closed 2020-01-05 by SteinwayCaiZhuo

smallduiyue8, 2020-08-05:
How to solve this problem？thx

SteinwayCaiZhuo, 2020-08-05:
板子上少个跳线帽
missing a jumper cap on the board

SteinwayCaiZhuo, 2020-08-05:
> How to solve this problem？thx

板子上少了跳线帽
One less jumper cap on the board

smallduiyue8, 2020-08-05:
> > How to solve this problem？thx
> 
> 板子上少了跳线帽
> One less jumper cap on the board

老哥，具体是哪个位置啊

SteinwayCaiZhuo, 2020-08-05:
半年没碰，已经忘了，不好意思。

> > > How to solve this problem？thx
> > 
> > 
> > 板子上少了跳线帽
> > One less jumper cap on the board
> 
> 老哥，具体是哪个位置啊

smallduiyue8, 2020-08-05:
> 半年没碰，已经忘了，不好意思。
> 
> > > > How to solve this problem？thx
> > > 
> > > 
> > > 板子上少了跳线帽
> > > One less jumper cap on the board
> > 
> > 
> > 老哥，具体是哪个位置啊

好的，感谢！

bianglibeyond, 2021-02-04:
老哥你最后解决了吗？怎么接线呢？求助，我用的A板

bianglibeyond, 2021-02-04:
> > 半年没碰，已经忘了，不好意思。
> > > > > How to solve this problem？thx
> > > > 
> > > > 
> > > > 板子上少了跳线帽
> > > > One less jumper cap on the board
> > > 
> > > 
> > > 老哥，具体是哪个位置啊
> 
> 好的，感谢！

老哥你最后解决了吗？怎么接线呢？求助，我用的A板