RoboMaster/RoboRTS #44: ubuntu 18 (melodic )make error

sheng00125, 2020-07-30:
roborts_common::ErrorInfo roborts_global_planner::AStarPlanner::Plan(const PoseStamped&, const PoseStamped&, std::vector<geometry_msgs::PoseStamped_<std::allocator<void> > >&)’:
/home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.cpp:86:52: error: call of overloaded ‘abs(unsigned int)’ is ambiguous
         unsigned int dist = abs(goal_x - tmp_goal_x) + abs(goal_y - tmp_goal_y);
                                                    ^
In file included from /usr/include/c++/7/cstdlib:75:0,
                 from /usr/include/c++/7/stdlib.h:36,
                 from /opt/ros/melodic/include/ros/platform.h:37,
                 from /opt/ros/melodic/include/ros/time.h:53,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.h:20,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.cpp:18:
/usr/include/stdlib.h:837:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/7/cstdlib:77:0,
                 from /usr/include/c++/7/stdlib.h:36,
                 from /opt/ros/melodic/include/ros/platform.h:37,
                 from /opt/ros/melodic/include/ros/time.h:53,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.h:20,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.cpp:18:
/usr/include/c++/7/bits/std_abs.h:56:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
/usr/include/c++/7/bits/std_abs.h:61:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/7/bits/std_abs.h:70:3: note: candidate: constexpr double std::abs(double)
   abs(double __x)
   ^~~
/usr/include/c++/7/bits/std_abs.h:74:3: note: candidate: constexpr float std::abs(float)
   abs(float __x)
   ^~~
/usr/include/c++/7/bits/std_abs.h:78:3: note: candidate: constexpr long double std::abs(long double)
   abs(long double __x)
   ^~~
/usr/include/c++/7/bits/std_abs.h:84:3: note: candidate: constexpr __int128 std::abs(__int128)
   abs(__GLIBCXX_TYPE_INT_N_0 __x) { return __x >= 0 ? __x : -__x; }
   ^~~
/usr/include/c++/7/bits/std_abs.h:102:3: note: candidate: constexpr __float128 std::abs(__float128)
   abs(__float128 __x)
   ^~~
/home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.cpp:86:79: error: call of overloaded ‘abs(unsigned int)’ is ambiguous
         unsigned int dist = abs(goal_x - tmp_goal_x) + abs(goal_y - tmp_goal_y);
                                                                               ^
In file included from /usr/include/c++/7/cstdlib:75:0,
                 from /usr/include/c++/7/stdlib.h:36,
                 from /opt/ros/melodic/include/ros/platform.h:37,
                 from /opt/ros/melodic/include/ros/time.h:53,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.h:20,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.cpp:18:
/usr/include/stdlib.h:837:12: note: candidate: int abs(int)
 extern int abs (int __x) __THROW __attribute__ ((__const__)) __wur;
            ^~~
In file included from /usr/include/c++/7/cstdlib:77:0,
                 from /usr/include/c++/7/stdlib.h:36,
                 from /opt/ros/melodic/include/ros/platform.h:37,
                 from /opt/ros/melodic/include/ros/time.h:53,
                 from /opt/ros/melodic/include/ros/ros.h:38,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.h:20,
                 from /home/sheng/guanfang_ws/src/roborts_planning/global_planner/a_star_planner/a_star_planner.cpp:18:
/usr/include/c++/7/bits/std_abs.h:56:3: note: candidate: long int std::abs(long int)
   abs(long __i) { return __builtin_labs(__i); }
   ^~~
/usr/include/c++/7/bits/std_abs.h:61:3: note: candidate: long long int std::abs(long long int)
   abs(long long __x) { return __builtin_llabs (__x); }
   ^~~
/usr/include/c++/7/bits/std_abs.h:70:3: note: candidate: constexpr double std::abs(double)
   abs(double __x)
[... 89 lines cut ...]
   ^~~
/usr/include/c++/7/bits/std_abs.h:84:3: note: candidate: constexpr __int128 std::abs(__int128)
   abs(__GLIBCXX_TYPE_INT_N_0 __x) { return __x >= 0 ? __x : -__x; }
   ^~~
/usr/include/c++/7/bits/std_abs.h:102:3: note: candidate: constexpr __float128 std::abs(__float128)
   abs(__float128 __x)
   ^~~
roborts_planning/global_planner/a_star_planner/CMakeFiles/a_star_planner.dir/build.make:93: recipe for target 'roborts_planning/global_planner/a_star_planner/CMakeFiles/a_star_planner.dir/a_star_planner.cpp.o' failed
make[2]: *** [roborts_planning/global_planner/a_star_planner/CMakeFiles/a_star_planner.dir/a_star_planner.cpp.o] Error 1
CMakeFiles/Makefile2:6011: recipe for target 'roborts_planning/global_planner/a_star_planner/CMakeFiles/a_star_planner.dir/all' failed
Makefile:140: recipe for target 'all' failed
Base path: /home/sheng/guanfang_ws
Source space: /home/sheng/guanfang_ws/src
Build space: /home/sheng/guanfang_ws/build
Devel space: /home/sheng/guanfang_ws/devel
Install space: /home/sheng/guanfang_ws/install
####
#### Running command: "cmake /home/sheng/guanfang_ws/src -DCMAKE_BUILD_TYPE=Debug -DCATKIN_DEVEL_PREFIX=/home/sheng/guanfang_ws/devel -DCMAKE_INSTALL_PREFIX=/home/sheng/guanfang_ws/install -G Unix Makefiles" in "/home/sheng/guanfang_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/sheng/guanfang_ws/build"
####
make[1]: *** [roborts_planning/global_planner/a_star_planner/CMakeFiles/a_star_planner.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed

XinZhangk, 2020-07-30:
The problem with abs is a common one, and as the error message has suggested, it is caused by ambiguity of function abs. Specifically, the compiler is confused about which versions of abs it should use. 
It knows various implementations of this function but none takes unsigned int as input type.
You can try to fix it by casting the arguments (e.g., goal_x - tmp_goal_x) of the function into float type.

sheng00125, 2020-08-03:
Thank you it has been solved by your solution .

charmyoung, 2020-11-11:
commit b6b8f81 has fixed compatibility with ros melodic

closed 2020-11-11 by charmyoung