RoboMaster/RoboRTS #45: Ubuntu 16+kinetic roborts_msgs/TwistAccel.h: 没有那个文件或目录 compilation terminated.

sheng00125, 2020-08-03:
sheng_ws/src/RoboRTS/roborts_planning/local_planner/include/local_planner/local_planner_base.h:67:37: fatal error: roborts_msgs/TwistAccel.h: 没有那个文件或目录
compilation terminated.
RoboRTS/roborts_planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/build.make:93: recipe for target 'RoboRTS/roborts_planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/src/teb_local_planner.cpp.o' failed
CMakeFiles/Makefile2:6188: recipe for target 'RoboRTS/roborts_planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/all' failed
Makefile:140: recipe for target 'all' failed
make[2]: *** [RoboRTS/roborts_planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/src/teb_local_planner.cpp.o] Error 1
make[1]: *** [RoboRTS/roborts_planning/local_planner/timed_elastic_band/CMakeFiles/timed_elastic_band.dir/all] Error 2
make: *** [all] Error 2
Base path: /home/ros/sheng_ws
Source space: /home/ros/sheng_ws/src
Build space: /home/ros/sheng_ws/build
Devel space: /home/ros/sheng_ws/devel
Install space: /home/ros/sheng_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ros/sheng_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ros/sheng_ws/build"
####
Invoking "make -j4 -l4" failed

sheng00125, 2020-08-03:
It was solved by executing"catkin_make roborts_msgs_generate_messages"

sheng00125, 2020-08-03:
home/ros/sheng_ws/devel/lib/libtimed_elastic_band.so：对‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(g2o::Solver*)’未定义的引用
collect2: error: ld returned 1 exit status
RoboRTS/roborts_planning/local_planner/CMakeFiles/local_planner_node.dir/build.make:680: recipe for target '/home/ros/sheng_ws/devel/lib/roborts_planning/local_planner_node' failed
CMakeFiles/Makefile2:6056: recipe for target 'RoboRTS/roborts_planning/local_planner/CMakeFiles/local_planner_node.dir/all' failed
Makefile:140: recipe for target 'all' failed
make[2]: *** [/home/ros/sheng_ws/devel/lib/roborts_planning/local_planner_node] Error 1
make[1]: *** [RoboRTS/roborts_planning/local_planner/CMakeFiles/local_planner_node.dir/all] Error 2
make: *** [all] Error 2
Base path: /home/ros/sheng_ws
Source space: /home/ros/sheng_ws/src
Build space: /home/ros/sheng_ws/build
Devel space: /home/ros/sheng_ws/devel
Install space: /home/ros/sheng_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/ros/sheng_ws/build"
####
####
#### Running command: "make -j4 -l4" in "/home/ros/sheng_ws/build"
####
Invoking "make -j4 -l4" failed

sheng00125, 2020-08-03:
https://github.com/RainerKuemmerle/g2o/releases/tag/20170730_git

sheng00125, 2020-08-03:
sudo wget https://gitlab.com/libeigen/eigen/-/archive/3.2.1/eigen-3.3.1.tar.gz

sheng00125, 2020-08-03:
需要用早一些版本的g2o 还有 3.3.1的 eigen库。
https://github.com/RainerKuemmerle/g2o/releases/tag/20170730_git
shttps://gitlab.com/libeigen/eigen/-/archive/3.2.1/eigen-3.3.1.tar.gz

XinZhangk, 2020-08-03:
Hi 
You might not be aware of this but I think somewhere on the road the developers updated the repository to ROS melodic.
Since ROS is famous backward incompatible I don't think reverting back to kinetic is a good idea, for the purpose of receiving future updates. But of course you have your own reason to use kinetic. 

Perhaps it is easier to ask the developers for a copy of an earlier version of RoboRTS and do a diff between the current and old version, rather than reverse engineering the changes they made in the update.

charmyoung, 2020-11-11:
commit b6b8f81 has fixed compatibility with ros melodic

closed 2020-11-11 by charmyoung