RoboMaster/RoboRTS #47: Use local planner for diff drive vehicle

mysterybc, 2020-10-29:
##### Can I use the local planner pack for my diff drive vehicle by modify some parameters?
##### And is this local planner pack special designed for omni driving vehicle?
I tested it in my stage simulation,and it seems turn too early before it reach the goal, and i assume it's trying to reach the goal by moving it toward y axis, but i have already set **max_vel_y** and **acc_lim_y** to zero, should i change other params?

**Thanks for your help in advance!!**

charmyoung, 2020-10-30:
TEB algorithm is used for the local planner and theoretically it is compatible for both omni and diff drving vehicle. For the project RoboRTS, only omni driving vehicle is tested. For more information, you can refer to https://github.com/rst-tu-dortmund/teb_local_planner

closed 2020-11-11 by charmyoung

mysterybc, 2020-11-11:
I'm sorry for forgetting to reply !And Really thanks for your help。
Anyway，I failed to fix this problem, and I'm managing to apply origin teb into my project! 
Thanks♪(･ω･)ﾉ