RoboMaster/RoboRTS #56: c++: internal compiler error: Segmentation fault (program cc1plus)

zhjpqq, 2021-09-01:
编译环境：
Ubuntu18.04,  ROS-Melodic,  OpenCV3.4.9,   CUDA10.2,  GCC/G++5.5/7.5
[image]

编译时候报错：c++: internal compiler error: Segmentation fault (program cc1plus)
[image]

已经尝试过的解决方法:
找到相关issues：https://github.com/RoboMaster/RoboRTS/issues/40，但报错位置不一样，无法解决。
另参考其他博文，增加SWAP2G也不能解决，将GCC/G++从7.5降级到5.5也不能解决。
https://blog.csdn.net/u013095333/article/details/94188743

补充说明如下：
因为跟官方的ros-kinetic和ubuntu16版本不一样，用sudo apt-get install -y ros-melodic-cv-bridge安装cv-bridge，
但编译RoboRTS时会opencv报错 "CV_FOURCC找不到标识符"。
所以在src下单独添加vision_opencv进行cv_bridge的编译，且单独编译vision_opencv包可以通过。

编译出错日志如下，internal compiler error的出错位置有好几个：
[image]
[image]
[image]

# 连续多次重复catkin_make，详细日志如下：
[caktin_make  第1次编译.txt](https://github.com/RoboMaster/RoboRTS/files/7087569/caktin_make_log_.1.txt)
[catkin_make  第2次编译.txt](https://github.com/RoboMaster/RoboRTS/files/7087570/catkin_make.2.txt)
[image]

wdczz, 2021-12-01:
linux查看修改线程默认栈空间大小（ulimit -s)，默认空间1024=1M，过小，改为10240=10M，可以编译。
查看ulimit -s
1. 修改配置文件
sudo vim /etc/security/limits.conf
2.
nvidia hard stack 10240
nvidia soft stack 10240
ubuntu hard stack 10240
ubuntu soft stack 10240
root hard stack 10240
root soft stack 10240
从1024改成10240（就是给他改大一点）
3.重启
4.编译
————————————By 武汉科技大学 ————————————
——————终于解决了—————— 也秃了 ———————————

yydc-0, 2022-08-16:
> linux查看修改线程默认栈空间大小（ulimit -s)，默认空间1024=1M，过小，改为10240=10M，可以编译。 查看ulimit -s
> 
> 1. 修改配置文件
>    sudo vim /etc/security/limits.conf
> 
> nvidia hard stack 10240 nvidia soft stack 10240 ubuntu hard stack 10240 ubuntu soft stack 10240 root hard stack 10240 root soft stack 10240 从1024改成10240（就是给他改大一点） 3.重启 4.编译 ————————————By 武汉科技大学 ———————————— ——————终于解决了—————— 也秃了 ———————————

Thank you so much. You saved my life.

Freedom-Guo, 2022-08-16:
您的邮件我已经收到，假期愉快！