RoboMaster/RoboRTS #58: catkin_make problem

HakureiWeideng, 2022-12-11:
When catkin_make, I got this:

```
/home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/src/teb_optimal.cpp: In member function ‘boost::shared_ptr<g2o::SparseOptimizer> roborts_local_planner::TebOptimal::InitOptimizer()’:
/home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/src/teb_optimal.cpp:139:117: error: no matching function for call to ‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::remove_reference<std::unique_ptr<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >, std::default_delete<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> > > >&>::type)’
  new g2o::OptimizationAlgorithmLevenberg(std::move(blockSolver));
                                                                ^
In file included from /home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/include/timed_elastic_band/teb_optimal.h:72:0,
                 from /home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/src/teb_optimal.cpp:56:
/opt/ros/kinetic/include/g2o/core/optimization_algorithm_levenberg.h:45:16: note: candidate: g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(g2o::Solver*)
       explicit OptimizationAlgorithmLevenberg(Solver* solver);
                ^
/opt/ros/kinetic/include/g2o/core/optimization_algorithm_levenberg.h:45:16: note:   no known conversion for argument 1 from ‘std::remove_reference<std::unique_ptr<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >, std::default_delete<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> > > >&>::type {aka std::unique_ptr<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >, std::default_delete<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> > > >}’ to ‘g2o::Solver*’
```

My system is `ubuntu-16.04.7`, ROS is Kinetic . 

The libg2o is : `2016.4.24-0xenial-20201017-070803+0000.` .

When I saw other issue and found the version this project uesd, but I got this `Version '2016.4.24-0xenial-20171104-173138-0800' for 'ros-kinetic-libg2o' was not found` .

Please help!

Freedom-Guo, 2022-12-11:
您的邮件我已经收到，假期愉快！

miss0609, 2023-03-07:
> When catkin_make, I got this:
>
> ```
> /home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/src/teb_optimal.cpp: In member function ‘boost::shared_ptr<g2o::SparseOptimizer> [...]

Is this problem solved？, I have also been troubled by this problem for a long time

Freedom-Guo, 2023-03-07:
您的邮件我已经收到，假期愉快！

miss0609, 2023-03-14:
> 

请问有什么方法可以解决吗

LDWJG, 2024-05-16:
> > 当catkin_make时，我得到了这个：
> > ```
> > /home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/src/teb_optimal.cpp: In member function ‘boost::shared_ptr<g2o::SparseOptimizer> roborts_local_planner::TebOptimal::InitOptimizer()’:
> > /home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/src/teb_optimal.cpp:139:117: error: no matching function for call to ‘g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(std::remove_reference<std::unique_ptr<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >, std::default_delete<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> > > >&>::type)’
> >   new g2o::OptimizationAlgorithmLevenberg(std::move(blockSolver));
> >                                                                 ^
> > In file included from /home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/include/timed_elastic_band/teb_optimal.h:72:0,
> >                  from /home/weideng/UGV_ws/src/RoboRTS/roborts_planning/local_planner/timed_elastic_band/src/teb_optimal.cpp:56:
> > /opt/ros/kinetic/include/g2o/core/optimization_algorithm_levenberg.h:45:16: note: candidate: g2o::OptimizationAlgorithmLevenberg::OptimizationAlgorithmLevenberg(g2o::Solver*)
> >        explicit OptimizationAlgorithmLevenberg(Solver* solver);
> >                 ^
> > /opt/ros/kinetic/include/g2o/core/optimization_algorithm_levenberg.h:45:16: note:   no known conversion for argument 1 from ‘std::remove_reference<std::unique_ptr<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >, std::default_delete<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> > > >&>::type {aka std::unique_ptr<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> >, std::default_delete<g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1> > > >}’ to ‘g2o::Solver*’
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > 我的系统是，ROS是Kinetic的。`ubuntu-16.04.7`
> > libg2o 是： 。`2016.4.24-0xenial-20201017-070803+0000.`
> > 当我看到其他问题并找到这个项目的版本时，但我得到了这个.`Version '2016.4.24-0xenial-20171104-173138-0800' for 'ros-kinetic-libg2o' was not found`
> > 请帮忙！
> 
> 这个问题解决了吗？，这个问题我也困扰了很久

你好解决了吗，我也困扰好久

Freedom-Guo, 2024-05-16:
您的邮件我已经收到，假期愉快！